WHILL/mrp-support #81: WHILL CRへの緊急停止ボタンの接続方法について

Euclid-7, 2025-01-12:
私はWHILL CR(2ではない方)の研究を行っています。
・WHILL社の公式ホームページに、緊急停止ボタンの取り付けに関する質問が掲載されており、それに対する回答として、RxDとTxDを分離する事により、緊急停止機能を備えた事があると書かれてありました。その緊急停止ボタンの具体的な型番を教えていただけませんでしょうか。もしそれが難しい場合、WHILL CRの緊急停止ボタンとして推奨される別の型番を教えていただきたいです。
・下記のハードウェア構成に対して、緊急停止ボタンの配線はどのように繋がると考えられますか。
[image]
[image]
[image]
[image]
・緊急停止ボタンの取り付けに関する質問は、このgithubのページ以外だと、どこに質問するのが適切でしょうか。

私はハードウェアの電気的な構成に関して全く詳しくないため、ぜひ教えていただきたく。

よろしくお願いいたします。

GeorgeMandokoro, 2025-01-21:
お問い合わせいただきありがとうございます。
回答可能な質問につきまして下記の通り回答させていただきます。

> ・WHILL社の公式ホームページに、緊急停止ボタンの取り付けに関する質問が掲載されており、それに対する回答として、RxDとTxDを分離する事により、緊急停止機能を備えた事があると書かれてありました。その緊急停止ボタンの具体的な型番を教えていただけませんでしょうか。もしそれが難しい場合、WHILL CRの緊急停止ボタンとして推奨される別の型番を教えていただきたいです。

見ていただいた[FAQ](https://faq.whill.inc/jp/s/jp/%E7%B7%8A%E6%80%A5%E5%81%9C%E6%AD%A2%E6%A9%9F%E8%83%BD%E3%82%92%E6%90%AD%E8%BC%89%E3%81%97%E3%81%9F%E3%81%84%E3%81%AE%E3%81%A7%E3%81%99%E3%81%8C)の記載内容は参考情報ですので、申し訳ございませんが、より詳細な情報はお伝えできかねます。

> ・下記のハードウェア構成に対して、緊急停止ボタンの配線はどのように繋がると考えられますか。

一般的な回答とさせていただきますが、「ユーザーのコントローラ」―「WHILL Model CR」間の、TxD線およびRxD線を切断できるように、スイッチ(緊急停止ボタン)を配置していただければよいと考えます。

> ・緊急停止ボタンの取り付けに関する質問は、このgithubのページ以外だと、どこに質問するのが適切でしょうか。

個別カスタムの相談等は、[こちら](https://file.notion.so/f/f/85788558-f711-4955-85b9-fd1c43ca8b6b/623bf2f8-e41a-47a4-a26d-d8f13836ccb8/742-00474_WHILL%E9%9B%BB%E5%8B%95%E3%83%A2%E3%83%93%E3%83%AA%E3%83%86%E3%82%A3%E3%83%97%E3%83%A9%E3%83%83%E3%83%88%E3%83%95%E3%82%A9%E3%83%BC%E3%83%A0%E3%81%AE%E3%82%A2%E3%83%95%E3%82%BF%E3%83%BC%E3%82%B5%E3%83%BC%E3%83%93%E3%82%B9%E3%81%AB%E9%96%A2%E3%81%99%E3%82%8B%E3%81%8A%E7%9F%A5%E3%82%89%E3%81%9B.pdf?table=block&id=318cc848-c87b-4447-a964-b7691fbfca31&spaceId=85788558-f711-4955-85b9-fd1c43ca8b6b&expirationTimestamp=1737453600000&signature=Q6mKXK30rvhHp7ixsemtccIxF1SbKHI4wV2P4LmH8z8&downloadName=742-00474+WHILL%E9%9B%BB%E5%8B%95%E3%83%A2%E3%83%93%E3%83%AA%E3%83%86%E3%82%A3%E3%83%97%E3%83%A9%E3%83%83%E3%83%88%E3%83%95%E3%82%A9%E3%83%BC%E3%83%A0%E3%81%AE%E3%82%A2%E3%83%95%E3%82%BF%E3%83%BC%E3%82%B5%E3%83%BC%E3%83%93%E3%82%B9%E3%81%AB%E9%96%A2%E3%81%99%E3%82%8B%E3%81%8A%E7%9F%A5%E3%82%89%E3%81%9B.pdf)のアフターサービスをご検討いただきますようよろしくお願いいたします。

以上、よろしくお願いいたします。

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro