WHILL/mrp-support #85: Whill model cr2にて取得可能なデータについて

Yukihi-7777, 2025-07-17:
お世話になっております。
前回質問時にご丁寧な対応をしていただき誠にありがとうございます。

再度質問させていただけないでしょうか。
whill model cr2ではジョイスティックの値を取得できないのでしょうか。
model cr2にてros topicを確認したところ、ジョイスティックの数値が0のままであり、変化しないことに気づきました。ただし、オドメトリ等の数値は変化していました。
同様の確認をPythonや別のcr connect checkerを使用しても値は取れませんでした。

何か問題、またはコードの一部に問題があるのでしょうか。

思い当たる節がありましたら、教えていただけないでしょうか。

Yukihi-7777, 2025-07-17:
ご返信いただき、誠にありがとうございます。

おっしゃる通り、
「WHILL Model CR2本体に付属している物理ジョイスティックの値を取得したい」
という内容で相違ございません。

現在、自動運転のみならず、搭乗者が介入する場面における運転アシストの研究も進めております。
その一環として、本体付属のジョイスティックから得られる入力値を、搭乗者側の操作としてアシスト機能に活用しようと考えておりました。

ご案内いただいた通信仕様書も確認しておりましたが、従来のCRモデルでは取得が可能であったため、CR2でも何らかの方法があるのではないかと考え、ご質問させていただいた次第です。
お手数をお掛けしてしまい、大変申し訳ございません。

外部ジョイスティックを新たに取り付ける形で対応させていただきます。

お忙しい中、ご丁寧にご対応いただき、心より御礼申し上げます。

GeorgeMandokoro, 2025-07-18:
ご回答いただきありがとうございます。

ジョイスティックの値を取得できるようにしたいとの要望が他にもあったため、次回の全ユーザーに対するファームウェア配信時（詳細未定）に対応しようと考えておりました。

ですが、ご所望でしたら、ジョイスティックの値を取得できるようにした特殊ファームウェアを、先行配信という形で、個別に提供させていただきます。
希望する場合は、メールにて必要な情報（機体のシリアル番号など）をやり取りさせていただきたいと思いますが、いかがいたしましょうか？

Yukihi-7777, 2025-07-18:
ご案内いただき、誠にありがとうございます。
ぜひ、先行配信の特殊ファームウェアを希望いたします。

こちらのスレッドコメントから私のメールアドレスをお送りする形でよろしいでしょうか。
お手数をおかけいたしますが、何卒よろしくお願いいたします。

GeorgeMandokoro, 2025-07-18:
ファームウェア配信希望の旨、承知いたしました。

それでは、こちらの担当直通メールアドレス（mrp.contact@whill.inc）へ、
ジョイスティック値取得可能なCR2用の特殊ファームウェア配信希望の旨をまずお伝えください。

以降は、メールにてやり取りさせていただきますので、よろしくお願いいたします。

closed 2025-07-18 by Yukihi-7777